lgsvl/simulator #2018: Docker API responded with status code=NotFound

hailonghu, 2022-03-08:
Running VSE simulation failed, Docker API responded with status code=NotFound, response={"message":"No such image: wise.svlsimulator.com/svlcontent/runner/vse:latest"}
[image]

wannastar, 2022-03-08:
you failed to download the resources of vse or python-api docker. you can use your hotspot to do this again. if your failed again, your can contact me with my email 18811123155@163.com.

hailonghu, 2022-03-09:
> you failed to download the resources of vse or python-api docker. you can use your hotspot to do this again. if your failed again, your can contact me with my email [18811123155@163.com](mailto:18811123155@163.com).

Thank you very much, LGSVL is working fine with the docker image you provided now.

closed 2022-03-09 by hailonghu

yangjiale1995, 2022-03-17:
> > you failed to download the resources of vse or python-api docker. you can use your hotspot to do this again. if your failed again, your can contact me with my email [18811123155@163.com](mailto:18811123155@163.com).
> 
> Thank you very much, LGSVL is working fine with the docker image you provided now.

I have encountered the same problem. How can I solve it ?  My email is 13324520153@163.com,  ths .

abeer93m, 2022-05-04:
> python-api docker

Hello, I am facing the same issue. Could you please tell me how you solved it? I would really appreciate that. Thank you!

abeer93m, 2022-05-04:
> > > you failed to download the resources of vse or python-api docker. you can use your hotspot to do this again. if your failed again, your can contact me with my email [18811123155@163.com](mailto:18811123155@163.com).
> > 
> > 
> > Thank you very much, LGSVL is working fine with the docker image you provided now.
> 
> I have encountered the same problem. How can I solve it ? My email is [13324520153@163.com](mailto:13324520153@163.com), ths .

Hello, did you manage to solve it? Thank you.

abeer93m, 2022-05-05:
> you failed to download the resources of vse or python-api docker. you can use your hotspot to do this again. if your failed again, your can contact me with my email [18811123155@163.com](mailto:18811123155@163.com).

Hello, I am facing the same issue, could you please provide me with the solution? Thank you so much.

wannastar, 2022-10-11:
从网易163邮箱发来的超大附件
vse.tar (274.76M, 2022年5月26日 15:10 到期)
下载
python-api.tar (274.69M, 2022年5月26日 15:16 到期)
下载