lgsvl/simulator #2020: Upload vehicle failed Bundle manifest file is not found

llu0120, 2022-03-08:
Hi, 
I tried to download the vehicle model from here 
[](https://content.lgsvlsimulator.com/vehicles/lincolnmkz2017/)

and after finished downloading, I tried to upload the asset bundle to the cloud by Add New in the Vehicles Library. However, it failed with the error below. 
[image]

Can anyone help? 

Thanks in advance.

jim18324, 2022-03-09:
In the website it said it only support 2020.6 version of the simulator, and it have some change when building vehicle from unity between 2020.6 to latest version. So i think you simply cant use it.

llu0120, 2022-03-09:
Is it possible to have this assets bundle usable in the version of 2021.3? I would like to make some custom modification of the sensors plugin and vehicle dynamics plugin with using this LincolnMKZ2017 model.

jim18324, 2022-03-11:
Sorry, i have no idea how you can do it.

carTloyal123, 2022-03-25:
What is the asset file you are trying to upload? You have to compile a vehicle asset in Unity before being able to upload it to WISE for use in LGSVL.