lgsvl/simulator #2030: can not launch LGSVL

201520906050, 2022-03-11:
when i clicked the icon to launch LGSVL, it appeared for just half second and the suddenly disappeared. And I tried to launch it by command `./simulator`,and the output is:

> Set current directory to /home/pami/svlsimulator-linux64-2021.1
> Found path: /home/pami/svlsimulator-linux64-2021.1/simulator

So I checked the log in` ~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log ` , and the output is:

> Mono path[0] = '/home/pami/Downloads/lgsvlsimulator-linux64-2020.06/simulator_Data/Managed'
> Mono config path = '/home/pami/Downloads/lgsvlsimulator-linux64-2020.06/simulator_Data/MonoBleedingEdge/etc'
> Preloaded 'liblaszip.so'
> Preloaded 'libpng.so'
> Preloaded 'libsqlite3.so'
> Preloaded 'libturbojpeg.so'
> Display 0 'C27R502 27"': 1920x1080 (primary device).
> Display 1 'DELL P2419H 24"': 1920x1080 (secondary device).
> Loading player data from /home/pami/Downloads/lgsvlsimulator-linux64-2020.06/simulator_Data/data.unity3d
> Desktop is 1920 x 1080 @ 60 Hz
> [Vulkan init] extensions: count=18
> [Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
> [Vulkan init] extensions: name=VK_KHR_display, enabled=1
> [Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
> [Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
> [Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
> [Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
> Vulkan detection: 0
> No supported renderers found, exiting 
> (Filename:  Line: 618)

I also typed `vulkaninfo` in terminal, errors like that:

> ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Could not get 'vkCreateInstance' via 'vk_icdGetInstanceProcAddr' for ICD libGLX_nvidia.so.0
> ERROR: [Loader Message] Code 0 : /usr/lib/i386-linux-gnu/libvulkan_intel.so: wrong ELF class: ELFCLASS32
> ERROR: [Loader Message] Code 0 : /usr/lib/i386-linux-gnu/libvulkan_radeon.so: wrong ELF class: ELFCLASS32
> Cannot create Vulkan instance.
> This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.
> /build/vulkan-tools-136mCR/vulkan-tools-1.1.126.0+dfsg1/vulkaninfo/vulkaninfo.h:297: failed with ERROR_INCOMPATIBLE_DRIVER
> 

the cpu is AMD® Ryzen 7 5800x 8-core processor × 16 , GPU is Nvidia GeForce GTX 1080Ti, driver version is 510.54

Looking forward to your reply for handling the urgent issue ! Thank u !

201520906050, 2022-03-28:
Hello, but i have already installed VulkanSDK through the instuctions and the output was like that:
`(base) pami@pami:~/svlsimulator-linux64-2021.1$ sudo apt install libvulkan1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libvulkan1 is already the newest version (1.1.126.0-2~gpu18.04.1).
0 upgraded, 0 newly installed, 0 to remove and 208 not upgraded.
`
`(base) pami@pami:~/svlsimulator-linux64-2021.1$ sudo apt remove mesa-vulkan-drivers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'mesa-vulkan-drivers' is not installed, so not removed. Did you mean 'mesa-vulkan-drivers:i386'?
0 upgraded, 0 newly installed, 0 to remove and 208 not upgraded.`

 i tried this several times and none of them worked , so confused

carTloyal123, 2022-03-30:
It says at the bottom of that message "Vulcan drivers not installed so not removed" which sounds like the driver was never installed correctly to begin with. I would look at the Vulcan support pages.