lgsvl/simulator #2032: How can we get the distance information of each obstacles from LGSVL simulator?

Deepak3994, 2022-03-15:


wannastar, 2022-03-16:
3D Ground Truth sensor will do you a favor

Deepak3994, 2022-03-16:
@wannastar Can you provide little more information on how we can use the data from 3D ground truth sensor for distance calculation?

Deepak3994, 2022-03-17:
> 3D Ground Truth sensor will do you a favor

@wannastar  Looking forward for your reply..

s95huang, 2022-03-18:
> @wannastar Can you provide little more information on how we can use the data from 3D ground truth sensor for distance calculation?

Add a 3D ground truth sensor with ROS bridge will publish an LGSVL detected objects type ros msg.
You can then subscribe to this topic using a simple script that will give your the distance, velocity of nearby objects.

Deepak3994, 2022-03-19:
@s95huang I have done aas per you have suggested, Below is my data from 3D ground truth sensor..

lgsvl_msgs.msg.Detection3DArray(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=1647531976, nanosec=141935360), frame_id=''), detections=[lgsvl_msgs.msg.Detection3D(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''), id=12, label='Pedestrian', score=1.0, bbox=lgsvl_msgs.msg.BoundingBox3D(position=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=17.529476165771484, y=-13.135026931762695, z=-0.39085689187049866), orientation=geometry_msgs.msg.Quaternion(x=1.4748309240530943e-06, y=-1.971658548427513e-06, z=0.8787134289741516, w=0.47734978795051575)), size=geometry_msgs.msg.Vector3(x=1.326209545135498, y=0.44951820373535156, z=1.3071844577789307)), velocity=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=0.9738802909851074, y=0.012942089699208736, z=-0.8539438247680664), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=220.83616638183594))), lgsvl_msgs.msg.Detection3D(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''), id=13, label='Pedestrian', score=1.0, bbox=lgsvl_msgs.msg.BoundingBox3D(position=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=32.17742156982422, y=-3.3550338745117188, z=-0.26378053426742554), orientation=geometry_msgs.msg.Quaternion(x=-4.792945560438966e-07, y=2.415127255517291e-06, z=-0.9986836314201355, w=-0.05129365995526314)), size=geometry_msgs.msg.Vector3(x=0.533416748046875, y=0.5435590744018555, z=1.8645706176757812)), velocity=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=-0.0003734136698767543, y=-0.00020606104226317257, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=310.09930419921875))), lgsvl_msgs.msg.Detection3D(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''), id=14, label='Pedestrian', score=1.0, bbox=lgsvl_msgs.msg.BoundingBox3D(position=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=26.501930236816406, y=-2.1101036071777344, z=-0.26377061009407043), orientation=geometry_msgs.msg.Quaternion(x=-4.2593120497258496e-07, y=2.4251073682535207e-06, z=-0.9995718002319336, w=-0.029262984171509743)), size=geometry_msgs.msg.Vector3(x=0.5716901421546936, y=0.5418331623077393, z=1.8305789232254028)), velocity=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=0.9995512962341309, y=-0.00030180704197846353, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=-313.8094482421875))), lgsvl_msgs.msg.Detection3D(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''), id=15, label='Pedestrian', score=1.0, bbox=lgsvl_msgs.msg.BoundingBox3D(position=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=18.30001449584961, y=-12.697541236877441, z=-0.41528668999671936), orientation=geometry_msgs.msg.Quaternion(x=-2.017702627199469e-06, y=1.4111836890151608e-06, z=-0.6852286458015442, w=-0.7283280491828918)), size=geometry_msgs.msg.Vector3(x=0.4935179054737091, y=0.49095800518989563, z=1.6848642826080322)), velocity=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=-9.557171870255843e-05, y=-0.0001906325778691098, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=157.18675231933594))), lgsvl_msgs.msg.Detection3D(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''), id=16, label='Pedestrian', score=1.0, bbox=lgsvl_msgs.msg.BoundingBox3D(position=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=38.38329315185547, y=-3.073719024658203, z=-0.263783723115921), orientation=geometry_msgs.msg.Quaternion(x=-4.448481831786921e-07, y=2.4217088139266707e-06, z=-0.9993129372596741, w=-0.0370645634829998)), size=geometry_msgs.msg.Vector3(x=1.087406873703003, y=0.4633291959762573, z=0.7874084115028381)), velocity=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=0.9993364214897156, y=-0.00024484648020006716, z=-5.9604644775390625e-06), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=312.94793701171875))), lgsvl_msgs.msg.Detection3D(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''), id=17, label='Pedestrian', score=1.0, bbox=lgsvl_msgs.msg.BoundingBox3D(position=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=18.744112014770508, y=-5.4722700119018555, z=-0.4151962995529175), orientation=geometry_msgs.msg.Quaternion(x=2.047788939307793e-06, y=-1.3671589158548159e-06, z=0.6692958474159241, w=0.7429960370063782)), size=geometry_msgs.msg.Vector3(x=1.0874079465866089, y=0.4633314609527588, z=0.7874084115028381)), velocity=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=0.9679616093635559, y=0.024257050827145576, z=-1.1920928955078125e-05), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=153.08621215820312))), lgsvl_msgs.msg.Detection3D(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''), id=18, label='Pedestrian', score=1.0, bbox=lgsvl_msgs.msg.BoundingBox3D(position=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=19.01321792602539, y=-5.0510640144348145, z=-0.35515546798706055), orientation=geometry_msgs.msg.Quaternion(x=2.0008828869322315e-06, y=-1.434932642041531e-06, z=0.693789005279541, w=0.7201784253120422)), size=geometry_msgs.msg.Vector3(x=0.47898486256599426, y=0.49714896082878113, z=1.8187717199325562)), velocity=geometry_msgs.msg.Twist(linear=geometry [... cut ...] rnion(x=-4.444866590347374e-07, y=2.4217752070399e-06, z=-0.9993184804916382, w=-0.03691538795828819)), size=geometry_msgs.msg.Vector3(x=0.46566811203956604, y=0.4470740556716919, z=1.7760220766067505)), velocity=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=0.9993317723274231, y=-0.00016175827477127314, z=-5.9604644775390625e-06), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=312.977783203125)))])

Kindly, let me know where is the distance of each object from the above message.

wannastar, 2022-03-21:
With the 3D Groud Truth sensor, you can get the coordinates(x,y,z) relative to ego. Then you can get the absolute distance by equation dis = sqrt(pow(x,2) + pow(y,2) + pow(z,2)). But it may have some deviation, cause the (x,y,z) is at the cencer of the box of the npc.

Deepak3994, 2022-03-21:
@wannastar  So, I need to use 

bbox=lgsvl_msgs.msg.BoundingBox3D(position=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=38.727420806884766, y=-3.738780975341797, z=-0.2637871205806732)

This information for the calculation of distance (x,y,z)?

Deepak3994, 2022-03-22:
@wannastar  @s95huang 

Kindly help...

closed 2022-03-23 by Deepak3994