lgsvl/simulator #2035: Running lgsvl simulator in linux

zyan1234, 2022-03-17:
Hi ,
I am trying to run lgsvl simulator in a linux based azure GPU machine with NVIDIA drivers. I am unable to run the simulator successfully.I am running the simulator with ./simulator command but  getting the Segmentation fault ( coredump) error when I try to run the simulator with ./simulator command. 
Then I searched for the coredump file and opened the files with gdb debugger ,it is showing a list of issues .unable to understand what are those errors and how to solve them.Attaching the screen shot for the same

[image]

Please help me to solve the above errors and run the simulator in linux machine.
Thank you

EricBoiseLGSVL, 2022-03-21:
Be sure you have the supported GPU for this.

zyan1234, 2022-03-22:
Hi @EricBoiseLGSVL 
Thank you for the reply .
Yes we do have the supported GPU .
We did use the same GPU in windows and was able to run the simulator properly .
But unable to run the simulator in linux machine with the same GPU support.
Why the simulator is crashing on execution from terminal with ./simulator command

EricBoiseLGSVL, 2022-03-22:
Sorry but this seems like a setup issue with your linux machine.  Not sure I can help other than looking to installing mes drivers, making sure the correct gpu is used and needed dependencies on linux are installed.  I can't really tell anything from the log you provided.  Can you list hardware specs so maybe others can see the issue?