lgsvl/simulator #2039: Cannot run VSE .json

ijklmnopqr, 2022-03-21:
Hello,

I am trying to run simulation using VSE, but I get below error.
[image]

Could anyone please tell me a solution?
I am using apollo 5.0 it works fine for random traffic.

wannastar, 2022-03-22:
you need to download the vse docker image firstly. it's a automatic process but you failed because the network problem.

ijklmnopqr, 2022-03-22:
So how can I resolve it?

wannastar, 2022-03-23:
use your hotpots to download it

Jorge2091, 2022-04-02:
i get the same exact problem. i am using linux, is that a command line, what is hotpots. is there a link to it?

ijklmnopqr, 2022-04-02:
I think it was hotspot. Just try using your mobile hotspot so as to download vse docker. It downloads automatically as soon as you run vse simulation.
Please do tell me if it works.
As I have computer setup with no wifi, I was not able to apply this solution.
Maybe you can also try using VPN.

Jorge2091, 2022-04-02:
i also dont have that solution, but i will try by using wire conenction through my phone and see. but i was able to get it working by going into older version of lgsvl. for example the 21.1 is working. i will try newer once i get back.

ijklmnopqr, 2022-04-04:
Okay I will check it

closed 2022-06-06 by ijklmnopqr