lgsvl/simulator #2041: Reference for camera distortion.

peter0617ku, 2022-03-28:
Hi, LGSVL team.

In the simulator's source code, there is a section describing camera distortion. Because part of the code cannot be understood, it is impossible to obtain all the parameters of the camera, including the camera's internal parameters, external parameters, and distortion parameters. The comments in the program contain a URL that appears to describe information about camera distortion or fisheye lenses. Unfortunately, the link has expired. The URL is https://wiki.lgsvl.com/display/AUT/Lens+Distortion. I would like to ask if the LGSVL team can provide this document for my reference.

Thanks in advance to your team.

---

[image]

rongguodong, 2022-03-28:
You can refer to this paper for meanings of a1-4 and b1-4:
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC4934233/

peter0617ku, 2022-03-29:
Wow! Thank you so much. 
Having this paper gave me a better understanding of how camera distortion works in the simulator.

By the way, another issue (#1625) explains the calculation of the effective FOV of a "fisheye" camera. May I know the origin of this formula?

`f' = f + 2 * asin(Xi * sin(0.5 * f))`

Thank you for taking the time out of your busy schedule to help!

rongguodong, 2022-03-29:
I don't know. You can ask the one who provide that formula.

peter0617ku, 2022-03-30:
OK, thanks for your help!

closed 2022-03-30 by peter0617ku