lgsvl/simulator #2047: lgsvl_bridge cannot worker?

hanyh1191, 2022-04-03:
Hello, Everything are OK before today.
Today I start "lgsvl_bridge" from bash command, then show "[INFO] [lgsvl-bridge]: Listening on port 9090", this all OK. When I click "Run Simulation" from WebUI and start simulation, Then ERROR "[ERROR] [lgsvl-bridge]: rcl_publisher_init failed: 103" Just like the picture, all other config or source "setup.bash" command normal. Why? Thanks.
[image]

hanyh1191, 2022-04-03:
yeah, how to say, I found my AD stack not care this ERROR, and still work OK though I don't know why cause error.

carTloyal123, 2022-05-19:
You probably did not add a topic name to one of your sensors when setting up the sensor configuration. Check each sensor you have and make sure it has a unique topic name.