lgsvl/simulator #2048: How to get CAN sensor data from LGSVL and send using CAN hardware

ijklmnopqr, 2022-04-06:
Hello,

How can I process CAN data, especially Vehicle speed from CAN sensor and send it through CAN hardware?
Are there any examples available on how to access sensor data from LGSVL?

carTloyal123, 2022-05-19:
The CAN bus sensor is not a true CAN device but rather a way to get vehicle data through the simulator to your bridge of some kind. You can post process the "CAN" data from the simulator into actual CAN signals, but it does not appear to be made for actual CAN hardware to interact with. For example, if you use a ROS2 bridge then the CAN sensor just publishes normal data to a ROS2 topic which has nothing to do with actual CAN protocol.

ijklmnopqr, 2022-06-20:
So, how can I enable the CAN bus sensor? I am using Apollo 5.0 where I tried to enable CAN, but it does not work.

closed 2023-02-09 by ijklmnopqr