lgsvl/simulator #2057: when Start Simulation cannot Load resource

hanyh1191, 2022-05-03:
When I click the "start Simulation" button In Simulator Web. then the Online Resource loading keeps stalling， cannot load Success and then shutdown, Why? Many Map resource load happened the same problem.
thanks.
[image]

hanyh1191, 2022-05-03:
And the simulator appear "Loading..." then nothing.
[image]
[image]

hanyh1191, 2022-05-04:
[image]

carTloyal123, 2022-05-19:
Do you have a clock sensor in your setup? And if you switch to offline mode and try to run the same simulation does that help?