lgsvl/simulator #2059: Unable to read archive entries: Bad archive

1Jyfc, 2022-05-12:
Hello SVL team!

I want to create some new scenes based on your open source scenes like [SanFrancisco](https://github.com/lgsvl/SanFrancisco). However when I build a map and upload to the server, it ran into an error of "Unable to read archive entries: Bad archive". Even if I never change anything and build the original project, it will cause the same problem. Could you give me some help?

I use the latest [simulator](https://github.com/lgsvl/simulator) and Unity Hub 3.1.2, and the version of editor is 2020.3.19f1. My system is Ubuntu 18.04.

1Jyfc, 2022-05-12:
Hello, I tried to use another device with Ubuntu 20.04, and it still cause this problem. So I think there is no system error. I wonder whether it can work on your device. Thanks a lot!

carTloyal123, 2022-05-18:
You probably are missing a step in the environment setup process. Just because you open an asset in Unity does not mean it is immediately ready to be built and used. It could be a naming issue, but I would have to see your workspace to help more. Is the environment in the External/Environments folder in your workspace?