lgsvl/simulator #2064: Consistent crash for version 2021.3 running online since yesterday

lintglitch, 2022-05-17:
Right now every online map crashes immediately on multiple tested systems on version 2021.3. To reproduce just try to run any map in the new version using the cloud interface.

This is likely a server-side issue. Offline scenarios still work as expected.

carTloyal123, 2022-05-18:
After the crash, can you open the simulator again and see what is in your cached assets? Is the track causing the crash there?

lintglitch, 2022-05-23:
Thank you, the question about the cache lead me down the right path. Seems the cache might have been corrupted. I managed to fix it by changing to a different SVL version. Removing the cache directory seems to have solved the issue even for 2021.3. Still, its interesting that this issue happened to me on two different machines

Unfortunately I cannot append Player.log since I removed the directory.

I'll mark this issue as solved

closed 2022-05-23 by lintglitch