lgsvl/simulator #2066: Adding map to SVL Simulator

koraykoca, 2022-05-19:
Hi, 

My SVL Simulator version is 2021.3. I want to add my own map in lanelet2 format to the simulator through browser, but I am getting "Type of Asset Bundle does not match the requested type" error. I tried to download the "SingleLaneRoad" map, which is compatible with the version 2021.3, from the SVL Simulator library in lanelet2 format and added it to the library. Likewise, I am getting the same error as well. I tried to download and upload it in different formats, the error still persists.

How can I add my lanelet2 format maps to the simulator?

carTloyal123, 2022-05-19:
You have to open the map in Unity and compile it as an Environment asset. Please look at this link for more info: https://www.svlsimulator.com/docs/simulation-content/add-new-map/

koraykoca, 2022-05-24:
Thank you for the answer. I would want to ask another question: When I want to install a map from the SVL Simulator, I can install lanelet2 format to get osm file for the map. However, for Autoware, I need also pcd and yaml file for the map to visualize it in rviz. How can I find them, specifically the point cloud data?