lgsvl/simulator #2067: WISE API assetGuid missmatch

YuqiHuai, 2022-05-24:
Hi! I have been trying to identify cached files for each plugin and I noticed some assetGuid mismatch.

For CyberRT plugin (id=ff4ba32f-8885-4847-9a8b-fd21e6603202), I get its assetGuid information from the following API: https://wise.svlsimulator.com/api/v1/plugins/ff4ba32f-8885-4847-9a8b-fd21e6603202
It says assetGuid = 6aca1444-9ee5-4b77-a1f6-6bbc3bfb18c0

However, for the Vehicle Configuration of LincolnMKZ, I see the assetGuid is fffac85a-b291-4614-9e82-68a51954306
[image]

What are the purposes of the 2 different assetGuid, and why is the API giving an incorrect assetGuid?

carTloyal123, 2022-05-29:
The bridge assetguid and the owner id and the vehicle id are not supposed to be the same. They are all different because they reference different organization schemes. The bridge assetGuid is referencing the actual unity asset guid for the bridge plugin you are using. The owner id is referencing the person who uploaded the plugin to WISE. The id of the bridge is referring to the current instance of the bridge you are using with specific settings. This is as far as I can tell but it is still unclear. @EricBoiseLGSVL may be able to help shed some light here.

YuqiHuai, 2022-05-30:
Hi @carTloyal123 Thanks for the comment. Sorry I used the wrong word in the post which might have caused confusion. Instead of "for the vehicle configuration" I should have said "in the vehicle configuration"

The screenshot I provided is not the vehicle assetGuid for LincolnMKZ. It is part of the sensor configuration for LincolnMKZ.

The plugin ID is the same as the link I provided, but the assetGuid used in Sensor Configuration is not the same as the one provided in the Cloud API.

EricBoiseLGSVL, 2022-05-31:
We had multiple ids for API so it can we can use one id to spawn the vehicle and sensor config.

YuqiHuai, 2022-06-01:
Thanks for the response! But why different assetGuid for the same plugin ID? Shouldnt assetGuid-pluginId be 1-to-1?

EricBoiseLGSVL, 2022-06-01:
I'd have to ask the backend team but I think it was legacy from how we stored vehicle guids.

YuqiHuai, 2022-06-02:
Thank you for asking for that!
I remember when I forced SVL to download the wrong asset it said something about the simulator version (got ....xxx.0 but expecting .....xxx.4). Is this the legacy part you are talking about?

EricBoiseLGSVL, 2022-06-03:
No, that is version control we have in the asset logic.  Unity doesn't support backwards compatibility with shaders/logic/etc. Please don't force old versions, it will fail sometimes even without errors

YuqiHuai, 2022-06-03:
I see. Thank you so much for the response.