lgsvl/simulator #2068: SVL simulator Real time video stream issue

Shant-H, 2022-05-30:
Hello Dears, How can I get the real time video stream from SVL simulator to use it on external monitor in addition to simulator executable ? 
Is there any Python API similar to video/camera sensor to provide the video stream data or UDP port to listed during runtime ?
Thanks in advance.

carTloyal123, 2022-06-15:
You can use the ros2 bridge or create your own bridge.

Shant-H, 2022-06-15:
Thanks for your reply @carTloyal123

I'm using cyberRT bridge with provided python API's to read sensors data and its working on the local host.
According to your feedback can I use two different bridges (CyberRT & ROS2) in parallel to get the video stream data ?