lgsvl/simulator #2069: SVL Simulator Offline Mode Issue

Shant-H, 2022-05-30:
Hello Dears,
I have the following issue in running Python API examples on windows 10 PC in the "Offline Mode":

When I run "API Only" simulation using "PyCharm" everything is working fine AS LONG AS I have an internet connection on the PC even in simulator's "Offline Mode" without using Web UI it works fine.

But, when I run the same simulation in "Offline Mode" WITHOUT internet connection on the PC I'm getting the following error:

Python API Quickstart #5: Ego vehicle driving in circle
Traceback (most recent call last):
  File "C:\Users\Shant\Envs\pythonapi\PythonAPI\quickstart\05-ego-drive-in-circle.py", line 55, in <module>
         ego = sim.add_agent(env.str("LGSVL__VEHICLE_0", lgsvl.wise.DefaultAssets.ego_lincoln2017mkz_apollo5), lgsvl.AgentType.EGO, state)
  File "C:\Users\Shant\Envs\pythonapi\PythonAPI\lgsvl\utils.py", line 23, in new_f
         return f(*args, **kwargs)
  File "C:\Users\Shant\Envs\pythonapi\PythonAPI\lgsvl\simulator.py", line 164, in add_agent
        uid = self.remote.command("simulator/add_agent", args)
  File "C:\Users\Shant\Envs\pythonapi\PythonAPI\lgsvl\remote.py", line 69, in command
    raise Exception(data["error"])
Exception: Could not find asset with Id 47b529db-0593-4908-b3e7-4b24a32a0f70 (An error occurred while sending the request)

Process finished with exit code 1

It seems that even "Offline Mode" needs for internet connection to run the simulation or something is wrong in my configurations ?

YuqiHuai, 2022-05-31:
`47b529db-0593-4908-b3e7-4b24a32a0f70` is the sensor configuration id for ego_lincoln2017mkz_apollo5.
I am not sure how the internals work but my guess is even SVL is supposed to be working offline it is trying to make an API request to WISE to gather the sensor configuratoins for ego_lincoln2017mkz_apollo5.

Shant-H, 2022-06-01:
Thanks for your reply YuqiHuai
I recently knew that even in offline mode python APIs are requesting remote communications to LGSVL.WISE cloud server which is not good to use the simulator as stand alone application.

YuqiHuai, 2022-06-03:
You are welcome! It is not Python API making that remote request. Python API tells SVL client to look for vehicle `47b529db...`, and SVL client makes a GET request to https://wise.svlsimulator.com/api/v1/vehicles/47b529db-0593-4908-b3e7-4b24a32a0f70 which will tell SVL the corresponding vehicle assetGuid (Lincoln MKZ) + sensor configuration. 

When you visit the endpoint it says not exist, that is because you need to add a valid SimId in the request header. The actual response is attached here
[image]

To use the simulator as a stand-alone application, I built this local SVL cloud https://github.com/YuqiHuai/SORA-SVL which will give SVL the information needed to run in no internet mode. But you still need internet to set things up initially.

Shant-H, 2022-06-06:
Thanks for your reply YuqiHuai

Can I use your uploaded code on Windows 10 without installing Docker ?

mslavescu, 2022-06-19:
@Shant-H in this video we showcase how to run OSSDC SIM without wise cloud in API mode:
https://m.youtube.com/watch?v=fU_C38WEwGw

Check the issues here for updates:
https://github.com/OSSDC/OSSDC-SIM/issues