lgsvl/simulator #2071: Local SVL cloud - Run SVL Simulator without wise.svlsimulator.com

YuqiHuai, 2022-05-31:
Hi All,
Here is my initial version of **local SVL cloud**
https://github.com/YuqiHuai/SORA-SVL

SVL Simulator has been a powerful tool for a lot of people and I want to make sure we will be able to continue using it in the future.
I plan to rebuild WISE locally therefore only require us to point client's cloud_url to localhost (by adding a config.yml file)
This approach might be more viable because user will still need an interface to create vehicle sensor configurations in the future.

I started a draft that allows API only simulations and in turn can run most of the examples in the quick start directory for LGSVL Python API (https://github.com/lgsvl/PythonAPI/tree/master/quickstart). This version includes all maps, plugins, vehicles (official + user uploaded) assets.

A lot has to be done to make this local SVL as capable as WISE is. For example, it needs to support adding new vehicle sensor configurations, adding new simulation templates (random traffic, python script, etc). I hope that we can all work together to build as much as possible before the WISE cloud is turned down.

mslavescu, 2022-06-02:
@YuqiHuai let's integrate this with OSSDC SIM, I just released here, a copy of LG SVL Simulator to further evolve this amazing project:
https://github.com/OSSDC/OSSDC-SIM

Will provide demos shortly.

YuqiHuai, 2022-06-02:
Hi @mslavescu ! Yes, that is a great idea! I did look at your approach and felt we are attacking the same problem from different angles. I probably do not have as much knowledge as you do in Unity, therefore decided to do this local cloud approach which in theory requires less modification of simulator code. There are a number of problems that we might face when integrating them together:

1. this local SVL cloud requires no change in SVL simulator, therefore your original commit "add support for no SVL cloud " might need to be reverted or changed in other ways.
2. Users must run local SVL on their own machine because I only implemented a draft that allows 1 client connection. More needs to be done to make this a public service. Alternatively, this local SVL cloud can redirect asset download requests to your server, and keep everything else local.

I am actually in the OSSDC Slack channel therefore we can also discuss more in detail there!

jimaldon, 2022-06-13:
Any update on which approach have you two decided to go ahead with?

YuqiHuai, 2022-06-17:
Not yet. Marius hasn't reached out to me, and I have been busy with my schoolwork recently. I plan to polish the local cloud version further when I have some more time in the future, since even a local version of SVL client still benefits from having some kind of interface.

mslavescu, 2022-06-19:
We just posted first OSSDC SIM demo video and simulation instructions, see details here:
https://github.com/OSSDC/OSSDC-SIM-Demos/discussions/2

@YuqiHuai the OSSDC SIM already supports local or remote assets.

The Wise cloud is very good for quickly creating configurations/simulations, but I would like to see if we can create them through Python API (may requires some updates to support all resource types) first and export them for ease of sharing.

YuqiHuai, 2022-06-19:
Sounds good!

mslavescu, 2022-06-26:
Check this comment on how to use other maps and vehicles configs in OSSDC SIM:

https://github.com/OSSDC/OSSDC-SIM/issues/3#issuecomment-1166593428