lgsvl/simulator #2072: How can I add a bridge client to a vehicle?

josh-wende, 2022-06-02:
Hi,

I'm trying to work on getting the simulator to use local files, instead of relying on the WISE web UI. When I run a simulation from the Python API, though, I get the error `Agent '...' is missing bridge client`. How can I add a bridge client to the vehicle? What should the vehicle's JSON (the one with the sensor configuration included) look like? Is the BridgeClient a Unity component that needs to be added to the vehicle before building?

Thanks in advance for any guidance.

ertauren, 2022-06-03:
I would suggest looking at this issue first:
https://github.com/lgsvl/simulator/issues/1994

Then if it isn't suited to you somehow:
Instead adding it as component, I would suggest adding "BridgeData" to vehicle at AddAgent.cs file and let the SVL handle the creating component part.
You will need to handle config object at :
https://github.com/lgsvl/simulator/blob/1d98de7a8388d1c364e518e8629fdbb77ce521a7/Assets/Scripts/Api/Commands/AddAgent.cs#L113
this is probably cause of the first add_agent command's error.
that config object initialized at :
https://github.com/lgsvl/simulator/blob/1d98de7a8388d1c364e518e8629fdbb77ce521a7/Assets/Scripts/Api/Commands/AddAgent.cs#L79
so in fact you need to initialize BridgeData of vehicleData object at that line.

I didn't really test it. If it doesn't work right away, I still suggest tweaking that vehicleData object to your needs, maybe initialize it at the beginning (ConnectionUI.cs)

josh-wende, 2022-06-03:
Thanks @ertauren! This was the correct line of inquiry. Putting the proper bridge info into the correct place in the vehicleData json got it working.

closed 2022-06-03 by josh-wende