lgsvl/simulator #2075: stuck on the offline mode

githjk, 2022-06-04:
When i open the simulator, it stucked on the offline mode. Click on the "LINK TO CLOUD"  button or the offline dropbox,are not response.My OS is Ubuntu20.04

[image]

[image]

Below is contents in Player.log
```shell
Mono path[0] = '/home/tong/Downloads/svlsimulator-linux64-2021.3/simulator_Data/Managed'
Mono config path = '/home/tong/Downloads/svlsimulator-linux64-2021.3/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'UnityFbxSdkNative.so'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 '0': 1920x1080 (primary device).
Loading player data from /home/tong/Downloads/svlsimulator-linux64-2021.3/simulator_Data/data.unity3d
Unable to load player prefs
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2020.3.3f1 (76626098c1c4)
[Subsystems] Discovering subsystems at path /home/tong/Downloads/svlsimulator-linux64-2021.3/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] SelectPhysicalDevice requestedDeviceIndex=-1 xrDevice=(nil)
[Vulkan init] Physical Device 0x1fd0930 [0]: "NVIDIA GeForce MX250" deviceType=2 vendorID=10de deviceID=1d13
[Vulkan init] Selected physical device 0x1fd0930
[Vulkan init] Graphics queue count=1
[Vulkan init] extensions: count=136
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_buffer_device_address, enabled=0
[... 151 lines cut ...]

ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
  at System.ThrowHelper.ThrowArgumentOutOfRangeException (System.ExceptionArgument argument, System.ExceptionResource resource) [0x00029] in <eae584ce26bc40229c1b1aa476bfa589>:0 
  at System.ThrowHelper.ThrowArgumentOutOfRangeException () [0x00000] in <eae584ce26bc40229c1b1aa476bfa589>:0 
  at CloudAPI.GetInfo () [0x001c4] in <33a3e48d8c314f54845dfb34c374a12a>:0 
  at Simulator.Web.ConnectionUI.Awake () [0x00019] in <33a3e48d8c314f54845dfb34c374a12a>:0 

Desktop is 1920 x 1080 @ 60 Hz
ArgumentNullException: Value cannot be null.
Parameter name: configuration
  at PetaPoco.Database..ctor (PetaPoco.IDatabaseBuildConfiguration configuration) [0x00016] in <d3306f055f6c490aada69d8080999d36>:0 
  at PetaPoco.DatabaseConfigurationExtensions.Create (PetaPoco.IDatabaseBuildConfiguration source) [0x00001] in <d3306f055f6c490aada69d8080999d36>:0 
  at Simulator.Database.DatabaseManager.Open () [0x00000] in <33a3e48d8c314f54845dfb34c374a12a>:0 
  at Simulator.Database.Services.ClientSettingsService.GetOrMake () [0x00000] in <33a3e48d8c314f54845dfb34c374a12a>:0 
  at ConnectionManager.Start () [0x00030] in <33a3e48d8c314f54845dfb34c374a12a>:0 

Setting up 4 worker threads for Enlighten.
  Thread -> id: 7f371ffff700 -> priority: 1 
  Thread -> id: 7f371f7fe700 -> priority: 1 
  Thread -> id: 7f371effd700 -> priority: 1 
  Thread -> id: 7f371e7fc700 -> priority: 1 

```
What should I do or check? Thank you.

carTloyal123, 2022-06-15:
Looks like a possible issue with the backend service not supplying the correct data to the simulator. Please update to simulator version 2021.3 if you have not. Also try to clear the cache of your install if you can.

ruilau, 2022-06-17:
Did you solve this problem? When I try to use the version of Deep Blue Academy or 2021.3, it is offline, and the offline button does not respond. ubuntu18

githjk, 2022-06-17:
I tried that,but it didn't work.Thans for your suggestions!<br/><br/><br/><html>
<head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
</head>
<body>
<div class="ntes-mailmaster-quote" style="padding-top: 1px; padding-bottom: 1px" >
    <div style=" margin-top: 20px; margin-bottom: 12px; font-size: 14px; line-height: 1.25; color: #89898c; " >---- Replied Message ----</div>
    <div style=" margin-bottom: 12px; font-size: 13px; line-height: 1.25; color: #2c2c2e; padding: 2px; border-radius: 8px; background-color: #f0f0f0; " >
        <table width="100%" cellpadding="0" cellspacing="10" border="0">
            <tr>
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; white-space: nowrap; " >From</td>
                <td valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " ><a class="mail-from" style="color: #1c83eb; text-decoration: none" ***@***.***" ***@***.***&gt;</a></td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; white-space: nowrap; " >Date</td>
                <td class="mail-date" valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " >06/15/2022 10:14</td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; " >To</td>
                <td valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " ><a class="mail-to" style="color: #1c83eb; text-decoration: none" ***@***.***" ***@***.***&gt;</a></td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; " >Cc</td>
                <td valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " ><a class="mail-cc" style="color: #1c83eb; text-decoration: none" ***@***.***" ***@***.***&gt;</a>、<a class="mail-cc" style="color: #1c83eb; text-decoration: none" ***@***.***" ***@***.***&gt;</a></td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; " >Subject</td>
                <td class="mail-subject" valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " >Re: [lgsvl/simulator] stuck on the offline mode (Issue #2075)</td>
            </tr>
        </table>
    </div>
    <div><p></p>
<p dir="auto">Looks like a possible issue with the backend service not supplying the correct data to the simulator. Please update to simulator version 2021.3 if you have not. Also try to clear the cache of your install if you can.</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">&mdash;<br />
<script type="application/ld+json">[
{
"target": "https://github.com/lgsvl/simulator/issues/2075#issuecomment-1155899389",
"url": "https://github.com/lgsvl/simulator/issues/2075#issuecomment-1155899389",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
</div>
</div>
</body>
</html>

githjk, 2022-06-17:
I delete all the things on the website such as clusters and simulators，the problem is solved. <br/><br/><br/><html>
<head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
</head>
<body>
<div class="ntes-mailmaster-quote" style="padding-top: 1px; padding-bottom: 1px" >
    <div style=" margin-top: 20px; margin-bottom: 12px; font-size: 14px; line-height: 1.25; color: #89898c; " >---- Replied Message ----</div>
    <div style=" margin-bottom: 12px; font-size: 13px; line-height: 1.25; color: #2c2c2e; padding: 2px; border-radius: 8px; background-color: #f0f0f0; " >
        <table width="100%" cellpadding="0" cellspacing="10" border="0">
            <tr>
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; white-space: nowrap; " >From</td>
                <td valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " ><a class="mail-from" style="color: #1c83eb; text-decoration: none" ***@***.***" ***@***.***&gt;</a></td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; white-space: nowrap; " >Date</td>
                <td class="mail-date" valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " >06/18/2022 00:46</td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; " >To</td>
                <td valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " ><a class="mail-to" style="color: #1c83eb; text-decoration: none" ***@***.***" ***@***.***&gt;</a></td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; " >Cc</td>
                <td valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " ><a class="mail-cc" style="color: #1c83eb; text-decoration: none" ***@***.***" ***@***.***&gt;</a>、<a class="mail-cc" style="color: #1c83eb; text-decoration: none" ***@***.***" ***@***.***&gt;</a></td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; " >Subject</td>
                <td class="mail-subject" valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " >Re: [lgsvl/simulator] stuck on the offline mode (Issue #2075)</td>
            </tr>
        </table>
    </div>
    <div><p></p>
<p dir="auto">Did you solve this problem? When I try to use the version of Deep Blue Academy or 2021.3, it is offline, and the offline button does not respond. ubuntu18</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">&mdash;<br />
<script type="application/ld+json">[
{
"target": "https://github.com/lgsvl/simulator/issues/2075#issuecomment-1159057572",
"url": "https://github.com/lgsvl/simulator/issues/2075#issuecomment-1159057572",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
</div>
</div>
</body>
</html>

carTloyal123, 2022-06-18:
Good! Feel free to close the issue, glad it is fixed. Probably an out of date asset or something similar.

closed 2022-06-18 by githjk

DOMINICQQQ, 2022-07-11:
Hi,i have the same problem,do I need to delete everything on the cloud?