lgsvl/simulator #2080: Unable to sign into SVL Simulator

darshanmat, 2022-06-22:
It gives an error at the right bottom of the page saying **_something is wrong please try later!_**

darshanmat, 2022-06-22:
Tried resetting the password and yet I get the same error

knnurl, 2022-06-22:
+++, not logging in from any device for a while now.

okuzucu20, 2022-06-22:
I am also facing with the same issue

mslavescu, 2022-06-22:
Check this issue for a fork of LG SVL which doesn't require wise cloud:
https://github.com/lgsvl/simulator/issues/1994

WZhijun154, 2022-06-23:
help!

JasonBourne1998, 2022-06-23:
I guess you could use #2071 to coutinue using SVL

YuqiHuai, 2022-06-23:
Yes. I collected all the publicly available assets and built this local WISE. You can keep your existing projects going if I was able to collect your sensor configuration.

OSSDC is actually modifying the client so improvements can be made. Right now both approaches lack ability to add private asset. I’m also working on this feature for the local cloud.

darshanmat, 2022-06-23:
I am able to sign in now(via cloud)

mslavescu, 2022-06-26:
Check this comment on how to use other maps and vehicles configs in OSSDC SIM:

https://github.com/OSSDC/OSSDC-SIM/issues/3#issuecomment-1166593428

zhanghe1876168, 2022-08-11:
> It gives an error at the right bottom of the page saying **_something is wrong please try later!_**

hello,have you solved this problem？