lgsvl/simulator #2088: How do you get the position change information of the simulated vehicle and the nearest NPC?

zhanghe1876168, 2022-08-04:
Hello,

When using the emulator, I want to solve the following problems:

1. There are created NPCS in the map, as well as simulated vehicles.

2. As the simulated vehicles and NPCS move, their relative positions change.

3. How do you get the position change information of the simulated vehicle and the nearest NPC? and how to output the information?

Timothyoung97, 2022-10-05:
Have you implemented it? Would like to know how to implement this as well! Thank you very much!

Wangd521, 2022-10-11:
有个传感器的  3DGroundTruthSensor

Muxindawang, 2022-11-21:
> 有个传感器的 3DGroundTruthSensor

3DGroundTruthSensor显示position的xyz是通过经纬度设置的，你的也是这样的吗，请问怎么更改为米作为单位呢？或者直接获取距离车辆的xyz距离，非常感谢

Wangd521, 2023-03-13:
MapOrigin  每个地图里都有这个记录地图信息  这个脚本里有提供转换坐标的方法 utm和unity世界坐标以及经纬度三个坐标系互相转换