lgsvl/simulator #2091: wise.svlsimulator.com is down

torgeros, 2022-08-11:
The WISE (web user interface) is not available right now. While <wise.svlsimulator.com> itself is still up, the login page for example returns `502` or `504` responses, making SVL essentially unusable.

As described in https://github.com/lgsvl/simulator/blob/4c342fd39c5eca6ce6538a144f51d5929812e417/README.md?plain=1#L28, WISE is only supposed to be working until at least 2022-06-30.

This Issue can be used to discuss updates on this topic.

H0xC0000005, 2022-08-12:
While I understand the decision that LG doesn't maintain this anymore, I would rather say LGSVL already has a community and it is such a legacy that many people use. This down actually at least affects me very much to run simulations and let's trace this more.

PamLiam, 2022-08-13:
I just know SVL and use it for one week. Please make it WISE online again.

eagafonovlge, 2022-08-16:
Hey! It's up and running again.

closed 2022-08-16 by eagafonovlge

sixiaixixiya, 2022-08-18:
Hey! Is anybody else met the LGSVL frequently lost links? I guess the server of LGSVL is till not stable!