lgsvl/simulator #2095: LGSVL Simulator's Error When Use VSE

sixiaixixiya, 2022-08-16:
Hi all my friedns! I meet the lgsvl simulator's error when i use VSE(web) to load my scenario of Json file saved by visual editor. Is anyone else met the same error? I really need your help!
[image]

sixiaixixiya, 2022-08-16:
Maybe the picture is not clear! Here is the main error's screenshot.
[image]

dongtaihong, 2022-11-07:
hey bro, did you solve this problem? I met this too, hope your help, thanks a lot!

sixiaixixiya, 2022-11-08:
> hey bro, did you solve this problem? I met this too, hope your help, thanks a lot!

I'm sorry that I didn't solve the problem. But I used the template of api-only to load my self-built scene.

blueclocker, 2022-11-17:
hi, i had the same problem when use ros/autoware bridge but successed with apollo, which bridge did you use?