lgsvl/simulator #2103: SVL Simulator unexpectedly closes right after initiating a simulation

YouBotica, 2022-09-19:
Hello!
I've been using SVL simulator for more than one year, and I have never had this problem before. Right now I am using a new computer, and when I start any simulation in SVL, the simulator closes and it shows the following log in the terminal: 

Set current directory to /home/bg/Desktop/svlsimulator-linux64-2021.3
Found path: /home/bg/Desktop/svlsimulator-linux64-2021.3/simulator
Segmentation fault (core dumped)

I can open the executable without any problem, but when I start a simulation, it unexpectedly closes. I already tried to remove mesa vulkan drivers, and still, the error persists.

nvidia-smi output is the following:

[image]

The Player.log contains the following:
[Player.log](https://github.com/lgsvl/simulator/files/9602405/Player.log)

Any ideas on what might be causing this issue? I am using Quadro P400 nvidia gpu.

Best regards.

YouBotica, 2022-11-05:
I was able to solve the issue, the problem was that nvidia Quadro is not enough for running SVL, I ran the command nvidia-smi -l 1 in order to loop the command every 1 second, and find out that the 2048 MB allocated memory overflowed when I started running the simulation. The solution is to run a lighter simulation, run it in headless mode, or simply, changing computer because this one falls short. Btw, it has nothing to do with the fact that SVL has stopped developing the simulator, it still works fine for me.

closed 2022-11-05 by YouBotica

lily410, 2023-04-02:
> I was able to solve the issue, the problem was that nvidia Quadro is not enough for running SVL, I ran the command nvidia-smi -l 1 in order to loop the command every 1 second, and find out that the [...]
 
hello @YouBotica ,i have the same problem with you.
"run it in headless mode" does it mean that the "headless" in config.yml file should be set to true?is there any difference after running? thanks

nvidia-smi output:
[image]