lgsvl/simulator #2114: cannot open web browser localhost:8080

aiden00713, 2022-12-02:
I can not open browser when I click the button "open browser ..." in the LGSVL simulator window.
[image]
[image]
How to solve this problem? It looks like a server problem.

filipthoen2, 2022-12-07:
If you open https://account.lgsvlsimulator.com/ in a web browser, the access will time out.

Looks like the new account access page is https://wise.lssvlsimulator.com/sign-in.

Not sure this helps,

Filip

aiden00713, 2022-12-08:
> If you open https://account.lgsvlsimulator.com/ in a web browser, the access will time out.
> 
> Looks like the new account access page is https://wise.lssvlsimulator.com/sign-in.
> 
> Not sure this helps,
> 
> Filip

Thanks for your reply! 

But my course homework is using an old version, it can't use the new webUI to sign in.

Xiaowuzjj, 2022-12-25:
Have you solved the problem？

aiden00713, 2022-12-25:
> 

No, I haven't solved this problem.

aiden00713, 2022-12-25:
> Have you solved the problem？

No, I haven't solved this problem.

Xiaowuzjj, 2022-12-27:
> > Have you solved the problem？
> 
> No, I haven't solved this problem.

Is it out of service?

aiden00713, 2022-12-27:
> > > Have you solved the problem？
> > 
> > 
> > No, I haven't solved this problem.
> 
> Is it out of service?

Maybe...I guess.

Farweizhi, 2023-08-31:
I had the same problem, did you solve yours?

aiden00713, 2023-08-31:
> I had the same problem, did you solve yours?

No, I haven't solved it yet...