lgsvl/simulator #2127: lgsvl service can't work

yaoweixiao-ux, 2023-02-06:
when I open lgsvl simulator, UI always show connecting to https://wise.svlsimulator.com
connection to cloud time out, giving up
Does lgsvl can't work now?

sixiaixixiya, 2023-02-06:
Yes, I've noticed that too. Is there any other way to build a scene offline？

zhuzhouzhou, 2023-02-06:
sad ~~~~ need a offline version

Fin-Del-Mundo, 2023-02-06:
maybe need to use another simulator like carla

sixiaixixiya, 2023-02-06:
> maybe need to use another simulator like carla

Does Carla have the same Scenario Editor like LGSVL?

Fin-Del-Mundo, 2023-02-06:
> 

svl base on unity, but carla base on unreal.

sixiaixixiya, 2023-02-06:
> > 
> 
> svl base on unity, but carla base on unreal.

I know it. But my question is that does Carla have the function of creating scenario by just dragging the obstacles in the simulation map like LGSVL's Visual Editor.

noorzzz, 2023-02-06:
Sad news :(

Does anyone any other simulator which supports Apollo ? I've tried some carla-apollo bridge but they lack traffic signs support. As for whether Carla has scenario editor , I doubt they have one, but do refer to their documentations.

YuqiHuai, 2023-02-07:
I started a project https://github.com/YuqiHuai/SORA-SVL back then to be prepared for this day. This is a local SVL server, it is basically an attempt to implement WISE, a.k.a. an offline version. I am very lucky to have many others helping to integrate features. I didn't get a chance to focus more on this project afterward, but maybe it makes sense to focus on it now.

I am also aware of another attempt to remove dependency from WISE https://github.com/OSSDC/OSSDC-SIM

yaoweixiao-ux, 2023-02-08:
perfect!!!!