lgsvl/simulator #2128: Cloud Service don't work

mehmetozturan, 2023-02-08:
Is any one know if they're going to fix this or not. I build my entire thesis on this simulator and know i can't reach it. It's been 3 days and i'm still waiting.

gxnse, 2023-02-08:
hahaha I'm kinda in the same situation. I'm doing my masters thesis and now it doesn't work. And I got a whole vehicle modelled and imported with dynamical characteristics, etc. Luckily I'm almost done with simulations and hopefully would finish the whole thing offline with what I got...

mehmetozturan, 2023-02-09:
> hahaha I'm kinda in the same situation. I'm doing my masters thesis and now it doesn't work. And I got a whole vehicle modelled and imported with dynamical characteristics, etc. Luckily I'm almost done with simulations and hopefully would finish the whole thing offline with what I got...

ahaha you're lucky, i'm also done with all the simulation and code things but i cleared the cache due to some problems and now i have to re-download all the configs&environments from the cloud :( i think i need to move all my system to a similar simulator and that is so much work...

YuqiHuai, 2023-02-14:
Checkout https://github.com/lgsvl/simulator/issues/2127#issuecomment-1420114500