lgsvl/simulator #2129: Does anyone snapshot the latest documentation of the svlsimulator?

umdreamer, 2023-02-13:
I found that the svlsimulator.com/docs has shut down. Does anyone snapshot the latest documentation of the svlsimulator?

gxnse, 2023-02-13:
You can use cache of some search engine. For instance, open _yandex.com_, copy _https://www.svlsimulator.com/docs/simulation-content/sensors-list/_ and then click _Saved Copy_ on the right appearing three dots

YuqiHuai, 2023-02-14:
I just manually created the source files for the LGSVL Python API documentation. https://github.com/YuqiHuai/SORA-SVL/tree/main/docs and created a temporary host for it https://www.ics.uci.edu/~yhuai/SORA-SVL/

If you have time and resources please consider contributing to the documentation! Or if you know how to automatically convert HTML back to Sphinx source code, please help 🥲

dfguooo, 2023-02-17:
I find these archived docs: https://web.archive.org/web/20220824070758/https://www.svlsimulator.com/docs/

lemketron, 2023-03-02:
> I find these archived docs: https://web.archive.org/web/20220824070758/https://www.svlsimulator.com/docs/

Yes, I seeded archive.org with the SVL Simulator doc URL so it would archive them.  :-) 

The older LGSVL docs are archived here:
https://web.archive.org/web/20200608075624/https://www.lgsvlsimulator.com/docs/

AlexandrZabolotny, 2023-08-01:
Hello everyone
i am using an older version of simulator 2020.06.
It does not require a connection to the cloud server wise.svlsimulator.com, just to the localhost:8080.

After reinstalling this version, I cannot open the local host, because I am redirected to 
https://www.lgsvlsimulator.com/?returnUrl=http%3A%2F%2Flocalhost%3A8080%2F

Maybe anybody has ideas? Let me know