lgsvl/simulator #2142: Cannot open lgsvl simulator in ubuntu

Farweizhi, 2023-08-31:
I installed version 2020.06 of lgsvl in ubuntu and it worked fine before. But recently after I press open browser, the web page that pops up says connection refused. Refreshing and installing a new browser doesn't fix the problem.
[image]
[image]
I'm looking for advice, thank you very much!

wsqwsq547, 2023-09-16:
LGSVL always flashes back after downloading on Ubuntu, according to the instructions of the official website to do it will still flash back, download on Windows but can not be connected, I do not know what to do.

s95huang, 2023-11-10:
[LG has made the difficult decision to suspend active development of SVL Simulator, as of January 1, 2022.](https://github.com/lgsvl/simulator#lg-has-made-the-difficult-decision-to-suspend-active-development-of-svl-simulator-as-of-january-1-2022)